lhndo/LH-Stinger #6: Printer cad file is not opening

Shamoon78, 2023-10-16:
The printer cad file is broken canot be extracted

lhndo, 2023-10-16:
Hi, 
Is it the fusion or the step file?
Does it fail when extracting or when trying to open it in a CAD program? If so, in what program?

Shamoon78, 2023-10-17:
It is the step file and give me error when extracting the file

walterwissmann, 2023-10-17:
Hey @Shamoon78

I have no problem to extract the 'LH Stinger.step.zip' on Linux Mint.
See:
[image]

BR
Walter

lhndo, 2023-10-17:
> It is the step file and give me error when extracting the file

Can you try downloading the archive again with another browser? Maybe the download was corrupted. 
What program are you using to extract it, so I could try to reproduce the issue.

Shamoon78, 2023-10-17:
I am using windows 11 and using the built in software for windows 11 ,win
10 used to need a program to extract certain files but win 11 can extract
all kinds of files. This is the first issue for me but happy to give 7 zip
a try and see how it goes .

lhndo, 2023-10-17:
It should be fixed now

walterwissmann, 2023-10-17:
On Linux Mint / FreeCAD it runs ;-)

closed 2023-10-17 by lhndo