lhndo/LH-Stinger #15: Breakout PCB Part Sourcing

lhndo, 2024-02-12:


closed 2024-03-20 by lhndo

iNinja9k, 2024-12-14:
bro how can i get a board?

lhndo, 2024-12-14:
> bro how can i get a board?

Fysetc sells a kit pcb kit that includes the harnesses on their website. Ping me if you get it because there are some pins that need to be flipped depending on your board.

iNinja9k, 2024-12-14:
thanks for the info bro, appreciate it.
PDank Douglas Korytkowski

iNinja9k, 2024-12-15:
OMG 800 dollar kit geeze I don't know if the price to performance is worth
it even to build this now do you think I could source the parts cheaper or
use substitutes for some parts will a normal like big tree board work for
the breakout fans and such. I mean any raspberry pi will work for klipper.
PDank Douglas Korytkowski

lhndo, 2024-12-15:
I meant the breakout board which is 29.9 or something. There's a link for it in the Stinger BOM as well.