lhndo/LH-Stinger #21: NOT AN ISSUE - an idea / suggestion

billbrach, 2024-12-09:
Could this PCB design not be made directly on 3-4MM G10/FR4, and use that as the actual surface to print on ??

I am tired with every 3D printer that I've bought, the aluminum bed warping.  Almost always, they end up low in the middle.  If this PCB foil pattern could be directly etched on a correctly sized/thickness of G10, you could eliminate having to glue/stick this to the actual print bed.

Thanks for reading !!

lhndo, 2024-12-09:
Hi Bill, That is what the PRUSA beds are made of and you can check those out here: https://github.com/prusa3d/Heatbed_MK52_magnetic/ . There are many aftermarket sources where you can buy those directly, in mk3/4 size or mini 

 I have looked into FR4, but in the LH Stinger case it's way too bendy for the indented use. The CFRP build plate forms a rigid assembly with the carriage providing a very sturdy platform. To some degree both the FR4 and the CFRP can bend and the final flatness is dependent on how the entire assembly is tuned against the rail carts. Depending on that it can vary between 1mm to 0.08mm ranges. The surface is very stable when heated.

closed 2024-12-13 by lhndo

billbrach, 2024-12-13:
Thanks, I'm looking for bigger beds, in the Ender 3 size, 235 x 235.  It seems to me, as the folks at JLCPCB, could print this directly on either aluminum, or thicker G10/FR4, that that might be an option.

BTW, how did you create the back and forth pattern of the traces on the PCB film that you did ?? Is that all done by hand, or is there a way to lay those down.  Cover the entire surface in copper, then remove areas to make the traces ??  I'm no novice to EasyEDA, having done 25 different PCB designs now, over the last 3 years, and working on more !!

lhndo, 2024-12-13:
You can epoxy, tape or glue the FR4 heater on whatever surface you want, so you don't have to get something done though an expensive process. For example, I believe that Prusa sacrifices one of the bed PCBs in which they route slots for magnets, and then they glue it together with the actual working heater PCB. This way they have a sandwich of 2 FR4 plates with a larger end thickness and stiffness. 

The circuit is just made with thick traces done manually in KiCad after calculating the required length and width. You can use the excel file included here: https://github.com/lhndo/LH-Stinger/tree/main/PCB/PI%20Bed%20Heater%20-%20200mm, which was done by [Róbert Lőrincz](https://www.orbiterprojects.com/). I picked it from his story : https://www.orbiterprojects.com/stories/story-of-the-orbitron-heated-bed-design/ . Takes a bit of time but it's not that difficult in the end. 

I have ported the project to EasyEDA and you can check it out here: https://oshwlab.com/andrei6/lhs-pi-bed-heater-200x200mm