lhndo/LH-Stinger #24: gcode macro SP_FORCE_SET_LANE not defined

Bad-Hackerman, 2025-01-24:
Steps to reproduce: have the current lane not defined an try to manually override it

lhndo, 2025-01-25:
Hi, can you please give me the command and parameters along with the exact error it triggers?
Is the rest of your mmu functional, as in you have done multi color prints with it?

Bad-Hackerman, 2025-01-25:
So here is the console after selectign Tool 0:
```
$ T0
echo: SP: Sidestep Right
!! SP: Unknown current loaded LANE -1. Use SP_FORCE_SET_LANE LANE=<0...>, SP_CLEAR_ERROR_CODES, SP_UNLOAD_LANE, SP_LOAD_LANE and finally RESUME to manually change the filament. 
!! SP: Print PAUSED and filament change interrupted due to previous errors. Please fix the issues before resuming
!! SP: PAUSED. Unloading LANE -1 >>>> Pending LANE 0
$ SP_FORCE_SET_LANE LANE=0
// Unknown `command:"SP_FORCE_SET_LANE"
```

While writing this I notice that the `SP_FORCE_SET_LOADED_LANE` command seems to do what `SP_FORCE_SET_LANE LANE` seems to need to do.

I haven't gotten it to work yet, just installed it yesterday, but i don't know if it is user error or a bug, but no matter the hub to sensor variable length, the filament only moves by around 20mm. When manually moving the filament with `SP_FORCE_MOVE_LANE` it moves the correct amount.

lhndo, 2025-01-25:
* SP_FORCE_SET_LANE was a typo, now it's only SP_FORCE_SET_LOADED_LANE

* I have submitted a fix that should load a lane even if nothing was loaded before

* Hub to sensor variable is just as a guide length, Moves that end up with a sensor check (homing moves) end up when the status of the sensor has changed, or when the move distance was complete

I would recommend going one by one though the macros from the guide to check that everything is operational and loads/unloads correctly  before printing

Bad-Hackerman, 2025-01-25:
I forgot to change the motor endstop pin.

closed 2025-01-25 by Bad-Hackerman