lhndo/LH-Stinger #27: Any alternative for servo motor for PICO MMU

luanft, 2025-02-15:
Hi,

I'm working to build the PICO mmu. I just found that the EMAX ES3004 Servo is not available at my location. 
Can I use these servos as alternative for the PICO mmu?
- Servo MG995
- Servo SG90
- Servo MG90S 360 
- Servo Emax ES08MA

I'm not familiar with the servo motor. Which one can work with the PICO mmu? 360 rotation servo or 180 servo is enough?

Thank you for your help

lhndo, 2025-02-15:
Hi,

360 degrees servos are for speed controlled movement, not angle control. so they can't be used in this application.  

Out of those, only the MG995 can be considered (180deg version) but you will need to consider the power delivery since the stall current draw is bigger than what a EBB 42 can supply, and you will fry the board.

The are have been some alternative servo mounts posted on the Discord server for standard size servos such as the MG995

I recommend reading this: https://github.com/lhndo/LH-Stinger/wiki/Pico-MMU#q-can-i-use-servo-x-instead

luanft, 2025-02-17:
Thank you. Seem the mount for MG995 is not available. I will design one for me and share it here.

lhndo, 2025-02-17:
Try this one: https://www.printables.com/model/1129133-stringer-pico-mmu-with-996r-servo

luanft, 2025-02-17:
This one fit to the MG995. Thank you

closed 2025-02-17 by luanft