lhndo/LH-Stinger #39: [PICO MMU] Sync move extruder and sp_motor do not move at the same time

dacan2810, 2025-08-24:
Klipper version: v0.13.0-229-g3aadda6f
Pico MMU config 0.9.2
I was inspected that now when unload. extruder and sp motor move in order ( sp motor grab filamane then stop, then extruder unload filament).
=> Sync move fail when unload

i see error at code below. when i switched to legacy mode. extruder and sp motor move togeter
when normal it move in order => fail which mean _SP_MOTOR  and FORCE_MOVE not working same time
I think because klipper changed. because i switch back to 0.9.1 (which is already work for me before) and it still happaned

     {% if legacy == 1 %}
        M83
        G92 E0
        _SP_MOTOR STOP=0 SYNC=0 SPEED={sync_speed} ACCEL={sp.motor_accel} DIST={sync_dist}
        G0 E{sync_dist} F{60*sync_speed}
      {% else %}
        _SP_MOTOR STOP=0 SYNC=0 SPEED={sync_speed} ACCEL={sp.motor_accel} DIST={sync_dist}
        FORCE_MOVE STEPPER=extruder VELOCITY={sync_speed} ACCEL={sp.motor_accel} DISTANCE={sync_dist}
      {% endif %}

lhndo, 2025-08-24:
Hi @dacan2810, thanks for the report!
I assume that's that the latest version to date

If so, it's very likely that the latest klipper changes also broke the way the sync manual stepper mode works, in particular to this command:

`MANUAL_STEPPER STEPPER=sp_motor ENABLE=1 SET_POSITION=0 SPEED={speed} ACCEL={accel} MOVE={dist} SYNC=1`
   
There was an attempt to fix the manual stepper in, but it addressed a more general issue: https://github.com/Klipper3d/klipper/commit/91b5e8e9425b5063c8a5bab9358d4c45c0e3430c 

You can try rolling back to this cl:

```
cd ~/klipper
sudo systemctl stop klipper
git reset --hard 17ce45d2
sudo systemctl start klipper
```

dacan2810, 2025-08-24:
Hi @lhndo. Thanks for reply
I can confirm it happened on lasted klipper.
Following your guide, i rollback to 17ce45d2 and it work fine with non-legacy mode.
As soon as update to lasted, it still happen.
It take me days try to fixing hotend and debug :D

closed 2025-11-22 by lhndo